intel/linux-npu-driver #16: CMake Error at third_party/cmake/vpux_elf.cmake:23 (add_subdirectory):   add_subdirectory given source "vpux_elf/vpux_elf" which is not an existing   directory. Call Stack (most recent call first):   third_party/CMakeLists.txt:7 (include)

bhardwaj-nakul, 2024-01-16:
-- Using CMAKE_BUILD_TYPE=Release
-- option ENABLE_OPENVINO_PACKAGE: OFF
-- option ENABLE_VPUX_COMPILER_BUILD: ON
-- ENABLE_SANITIZER not specified, using no sanitizers
-- option ENABLE_SANITIZER: 
```
CMake Error at third_party/cmake/vpux_elf.cmake:23 (add_subdirectory):
  add_subdirectory given source "vpux_elf/vpux_elf" which is not an existing
  directory.
Call Stack (most recent call first):
  third_party/CMakeLists.txt:7 (include)

CMake Error at third_party/cmake/vpux_elf.cmake:27 (set_property):
  set_property could not find TARGET vpux_elf.  Perhaps it has not yet been
  created.
Call Stack (most recent call first):
  third_party/CMakeLists.txt:7 (include)

```

-- Could NOT find LevelZero (missing: LevelZero_INCLUDE_DIR LevelZero_LIBRARY) 
-- LevelZero_LIBRARIES: 
-- LevelZero_INCLUDE_DIRS: 
-- LevelZero not found in the system, take one from third_party/level_zero
```
CMake Error at third_party/cmake/level-zero.cmake:18 (add_subdirectory):
  The source directory

    /home/cbrd/workbench/linux-npu-driver/third_party/level-zero

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  third_party/CMakeLists.txt:8 (include)

CMake Error at third_party/cmake/level-zero.cmake:21 (add_dependencies):
  Cannot add target-level dependencies to non-existent target "ze_loader".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.
Call Stack (most recent call first):
  third_party/CMakeLists.txt:8 (include)

CMake Error at third_party/cmake/level-zero.cmake:22 (install):
  install TARGETS given target "ze_loader" which does not exist.
Call Stack (most recent call first):
  third_party/CMakeLists.txt:8 (include)

CMake Error at third_party/CMakeLists.txt:12 (add_subdirectory):
  The source directory

    /home/cbrd/workbench/linux-npu-driver/third_party/googletest

  does not contain a CMakeLists.txt file.

CMake Error at third_party/CMakeLists.txt:16 (add_subdirectory):
  The source directory

    /home/cbrd/workbench/linux-npu-driver/third_party/yaml-cpp

  does not contain a CMakeLists.txt file.
```
-- Level Zero driver version: 1.1.0
-- LEVEL_ZERO_LIB_NAME = ze_loader
-- NAME = ubuntu
-- VERSION_ID = 22.04
-- Configuring incomplete, errors occurred!

jwludzik, 2024-01-19:
Hi, please, first download all git submodule by calling:
```
git submodule update --init --recursive
```

And then try to compile the driver

bhardwaj-nakul, 2024-01-20:
Thank you! This resolved the issue.

closed 2024-01-20 by bhardwaj-nakul