intel/linux-npu-driver #17: linux_npu_driver for generic linux, not resytricted by Ubuntu only

olegmikul, 2024-01-16:
I am not sure if it an issue or a "feature", so will start discussion here.

Can a driver be part of a generic linux core, not tight to Ubuntu only? Some other distros have some compelling features for meteor lake based laptops, so restricting a driver to Ubuntu only will severely restrict NPU based advantaged of such laptops.

Thanks

jwludzik, 2024-01-19:
The plan is to have a generic Linux driver as much as possible. Today our testing scope is narrowed to Ubuntu22.04. In future the number of targeted systems will be extended.

JohnAZoidberg, 2024-04-09:
I'm using this repo just fine on Fedora 40 Beta.

But to do anything useful you need https://github.com/intel/intel-npu-acceleration-library which is really not supported on non-Ubuntu.

jo-m, 2024-08-09:
What is the difference between this repo, and the ivpu driver which [is already upstreamed](https://github.com/torvalds/linux/tree/ee9a43b7cfe2d8a3520335fea7d8ce71b8cabd9d/drivers/accel/ivpu)?

And, why does the upstream kernel have the ivpu driver, but seemingly [no firmware for it is in linux-firmware](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/intel)?

jwludzik, 2024-08-09:
> What is the difference between this repo, and the ivpu driver which [is already upstreamed](https://github.com/torvalds/linux/tree/ee9a43b7cfe2d8a3520335fea7d8ce71b8cabd9d/drivers/accel/ivpu)?
>
> And, [...]

Please, familiarize yourself with README.md and docs/overview.md in linux-npu-driver repository. It explains the software stack. The user-space part is within linux-npu-driver, npu_plugin, level-zero and OpenVINO repositories

The firmware is not in linux-firmware because today the firmware is strictly connected with NPU compiler

ulissesf, 2025-02-28:
@jwludzik The firmware could still go to linux-firmware. Are you saying we don't have a way to match it properly with compiler?

I believe if more of this stack was upstream then it'd be easier for users of different distributions to get it. I'm using Fedora 41 on my LNL and it's not easy to install and keep NPU stack updated, so ideally Fedora should just package it and ship it.

jwludzik, 2025-03-05:
@ulissesf , yes. Latest decision is to push firmware to linux-firmware

Today way to match firmware with compiler is by using release packages from Linux NPU driver

xanderlent, 2025-08-21:
As of last week Fedora rawhide (the future version 44) should include everything you need. Install `intel-npu-driver` and `intel-npu-compiler`, as well as `linux-firmware`, but you probably have that already.

You should be able to see the NPU device in the result list from openvino afterwards (you'll need the `python3-openvino` package for this):
```bash
python3 -c 'import openvino; print(openvino.Core().get_available_devices())'
```

jwludzik, 2025-08-25:
> > [@ulissesf](https://github.com/ulissesf) , yes. Latest decision is to push firmware to linux-firmware
> > Today way to match firmware with compiler is by using release packages from Linux NPU driver
> 
> Is there a timeline when the firmware is pushed to linux-firmware and the npu can be used directly on distros besides ubuntu without too much effort?

Yes, there is an agreement to update NPU firmware in linux-firmware following each new release published in the [linux-npu-driver/releases](https://github.com/intel/linux-npu-driver/releases)