intel/linux-npu-driver #28: Failed to compile network. L0 createGraph result: ZE_RESULT_ERROR_UNSUPPORTED_VERSION, code 0x78000002. ERROR! MAPPED_INFERENCE_VERSION is NOT compatible with the ELF Expected: 6.1.0 vs received: 7.0.0

bittervan, 2024-05-13:
I am using the up-to-date `linux-npu-driver` and `openvino` repo to build the software stack on Ubuntu 22.04. After installation, I ran tried to convert the resnet50 network according to the openvino tutorials.

```python3
import openvino as ov
core = ov.Core()
print(core.available_devices)
print(core.get_property("NPU", "FULL_DEVICE_NAME"))

from pathlib import Path
MODEL_DIRECTORY_PATH = Path("model")
MODEL_DIRECTORY_PATH.mkdir(exist_ok=True)
model_name = "resnet50"

from torchvision.models import resnet50, ResNet50_Weights
pytorch_model = resnet50(weights=ResNet50_Weights.DEFAULT)
pytorch_model.eval()

precision = "FP16"
device = "NPU"
model_path = MODEL_DIRECTORY_PATH / "ir_model" / f"{model_name}_{precision.lower()}.xml"

model = None
if not model_path.exists():
    model = ov.convert_model(pytorch_model, input=[[1, 3, 224, 224]])
    ov.save_model(model, model_path, compress_to_fp16=(precision == "FP16"))
    print("IR model saved to {}".format(model_path))
else:
    print("Read IR model from {}".format(model_path))
    model = core.read_model(model_path)

compiled_model = core.compile_model(model, device)
```

But the program cannot run correctly.

```
['CPU', 'NPU']
Intel(R) AI Boost
Read IR model from model/ir_model/resnet50_fp16.xml
E: [unnamed] checkVersionCompatibility:38       ERROR! MAPPED_INFERENCE_VERSION is NOT compatible with the ELF Expected: 6.1.0 vs received: 7.0.0
Traceback (most recent call last):
  File "/home/bittervan/Codes/test/mnist.py", line 28, in <module>
    compiled_model = core.compile_model(model, device)
  File "/home/bittervan/Codes/openvino/bin/intel64/Release/python/openvino/runtime/ie_api.py", line 521, in compile_model
    super().compile_model(model, device_name, {} if config is None else config),
RuntimeError: Exception from src/inference/src/cpp/core.cpp:104:
Exception from src/inference/src/dev/plugin.cpp:54:
Exception from src/plugins/intel_npu/src/plugin/src/plugin.cpp:592:
Exception from src/plugins/intel_npu/src/plugin/src/compiled_model.cpp:61:
Check 'result == ZE_RESULT_SUCCESS' failed at src/plugins/intel_npu/src/compiler/src/zero_compiler_in_driver.cpp:716:
Failed to compile network. L0 createGraph result: ZE_RESULT_ERROR_UNSUPPORTED_VERSION, code 0x78000002. ERROR! MAPPED_INFERENCE_VERSION is NOT compatible with the ELF Expected: 6.1.0 vs received: 7.0.0
```

jwludzik, 2024-05-14:
Hi, it seems that the problem is with driver and compiler installed in system. Can you help me to understand what driver and compiler do you use? (if possible provide output from `dpkg -l | grep npu` command)

bittervan, 2024-05-14:
> Hi, it seems that the problem is with driver and compiler installed in system. Can you help me to understand what driver and compiler do you use? (if possible provide output from `dpkg -l | grep npu` command)

It seems that building the `linux-npu-driver` and the `openvino` repos directly won't result in an installed .deb package. I just exported the LD_LIBRARY_PATH to use the runtimes. Is this correct?

And how should I provide information about the two runtime libraries when they are not installed via deb packages, is there a way?

Thanks for your reply, and looking forward to your further suggestions.

jwludzik, 2024-05-14:
My assumption was that whenever you build from source you are responsible to install the libraries and track them. The CMake provides `install` option that copy the built files to system, but there is no option to remove installed files in CMake

> It seems that building the linux-npu-driver and the openvino repos directly won't result in an installed .deb package. I just exported the LD_LIBRARY_PATH to use the runtimes. Is this correct?

Yes, you are correct! The example is in README -> https://github.com/intel/linux-npu-driver?tab=readme-ov-file#building-a-driver-together-with-the-compiler

> And how should I provide information about the two runtime libraries when they are not installed via deb packages, is there a way?

I will investigate this topic and then I will give you a response.

Your main issue is in this line from logs:
```
ERROR! MAPPED_INFERENCE_VERSION is NOT compatible with the ELF Expected: 6.1.0 vs received: 7.0.0
```

The version incompatibility is caused because the installed firmware is older then compiler/driver. You can install the firmware manually (without cmake --install or deb packages):
```
# first make sure that firmware is pulled using Git LFS
git lfs pull
# copy the firmware 
sudo cp linux-vpu-driver/firmware/bin/vpu_37xx_v0.0.bin /lib/firmware/updates/intel/vpu/
# reload the driver
sudo rmmod intel_vpu; sudo modprobe intel_vpu
```

Please, check if this fix your issue. Thanks in advance

bittervan, 2024-05-14:
The problem is solved perfectly! I think I was using the outdated firmware provided by my Linux distro.

Thank you so much for your help!

closed 2024-05-14 by bittervan