intel/linux-npu-driver #31: Upgrading NPU from 1.1.0 -> 1.2.0 directions causes issues

ayf7, 2024-05-29:
When I run the suggested command:

dpkg --purge --force-remove-reinstreq intel-driver-compiler-npu intel-fw-npu intel-level-zero-npu level-zero

even after re-installing the packages as requested in the 1.2.0 release, it will not be able to detect ls/accel/accel0. In fact, on boot one of the errors that is thrown is

`  intel_vpu 0000:00:0b.0: enabling device (0000 -> 0002)
  intel_vpu 0000:00:0b.0: [drm] *ERROR* ivpu_fw_request(): Failed to request firmware: -2
  intel_vpu 0000:00:0b.0: [drm] *ERROR* ivpu_dev_init(): Failed to initialize firmware: -2
  intel_vpu 0000:00:0b.0: Failed to initialize VPU device: -2
  intel_vpu: probe of 0000:00:0b.0 failed with error -2`

Any pointers on how to retrieve this would be appreciated. Thank you!

jwludzik, 2024-06-03:
Based on the error from `dmesg`:
```
0000:00:0b.0: [drm] *ERROR* ivpu_fw_request(): Failed to request firmware: -2 
```

It seems that you are unable to install `intel-fw-npu_1.2.0.20240404-8553879914_ubuntu22.04_amd64.deb` or the firmware from the `intel-fw-npu` package has wrong name or firmware is installed in wrong directory

1. Can you review the log from `dpkg -i`? 
2. Can you provide kernel version?
3. Can you check if `vpu_37xx_v0.0.bin` is present in `/lib` (ex. `find /lib -name vpu_37xx_v0.0.bin`)?

I suspect that you are using an old kernel (ex. v6.4) that used `mtl_vpu_v0.0.bin` firmware name instead of `vpu_37xx_v0.0.bin`. In this case I would recommend to update a kernel or create a symbolic link to `vpu_37xx_v0.0.bin` with name `mtl_vpu_v0.0.bin` and reload kernel driver
```
cd /lib/firmware/updates/intel/vpu/
sudo ln -rsf vpu_37xx_v0.0.bin mtl_vpu_v0.0.bin
sudo rmmod intel_vpu
sudo modprobe intel_vpu
```

ayf7, 2024-06-03:
Thanks for the response - I was able to retrieve it somehow by reinstalling `linux-firmware` actually.

I am using linux kernel 6.5.0 with HWE. This is a brand new setup of Ubuntu 22.04, and 6.5 was provided upon installation. I'm not too sure how to upgrade to 6.8 kernel safely, or if it is even recommended.

jwludzik, 2024-06-04:
Using the old firmware from `linux-firmware` might be problematic. The inference workload might not work with linux-npu-driver 1.2.0. I would recommend to update the kernel, check our recommendation from release page -> https://kernel.ubuntu.com/mainline/v6.8.1/ 

Ubuntu Mainline kernel are built for latest Ubuntu distribution, but you can use them with older too. You should be able to install it (omit linux-headers as it requires new libc): 
* https://kernel.ubuntu.com/mainline/v6.8.1/amd64/linux-image-unsigned-6.8.1-060801-generic_6.8.1-060801.202403151937_amd64.deb
* https://kernel.ubuntu.com/mainline/v6.8.1/amd64/linux-modules-6.8.1-060801-generic_6.8.1-060801.202403151937_amd64.deb

If it is not what you need, then I recommend to move to Ubuntu24.04. The Ubuntu24.04 uses kernel v6.8

ayf7, 2024-06-04:
Yep, that seemed to work. Thank you!

closed 2024-06-04 by ayf7