intel/linux-npu-driver #54: The system cannot recognize NPU

RyanL00, 2024-10-11:
### Environment
OS:  Ubuntu 22.04.1
Kernel: 6.8.0-45-generic
CPU: Intel(R) Core(TM) Ultra 5 135H

###Issue
According to [https://github.com/intel/linux-npu-driver/releases/tag/v1.8.0](url) Installing the NPU driver,
when executing `ls /dev/accel/accel0`, it was found that there was no accel in/dev. 
`lspci | grep -i npu`  without output.

[image]

jwludzik, 2024-10-11:
Have you reboot the machine after installing driver? Can you check is intel_vpu kernel module successfully load in your system -
> https://github.com/intel/linux-npu-driver/blob/main/docs/overview.md#kernel-module-driver ?

RyanL00, 2024-10-12:
I have rebooted the machine, but it did not work. It was still found that there was no accel in /dev.
`modinfo intel_vpu` `lsmod | grep intel_vpu` `dmesg | tail -n 20`
output below:
[image]
[image]

jwludzik, 2024-10-14:
Sorry for late replay. Can you check the `dmesg | grep intel_vpu` and `lspci -d 8086:7d1d` commands and send output from them?

RyanL00, 2024-10-14:
I've tried rebooting the machine or reloading the device.
`dmesg | grep intel_vpu` and `lspci -d 8086:7d1d` without output.

yeah, it is the same issue [ openvinotoolkit/openvino#27002 ](url)

jlawryno, 2024-10-14:
You may have NPU disabled in BIOS. Please provide full output form `lspci -vvv` and check if NPU is enabled in advanced BIOS settings. If you cannot find NPU option in BIOS you may try resetting the BIOS settings to the default.

RyanL00, 2024-10-18:
> You may have NPU disabled in BIOS. Please provide full output form `lspci -vvv` and check if NPU is enabled in advanced BIOS settings. If you cannot find NPU option in BIOS you may try resetting the BIOS settings to the default.

Thanks, NPU is disabled in the BIOS, change it to enabled and the NPU will be recognized correctly.

closed 2024-10-18 by RyanL00