intel/linux-npu-driver #55: Issue compiling on Arch Linux

FabioLolix, 2024-10-19:
Hi, I'm trying to build this for curiosity, seems it don't like the lack of `VERSION_ID=`  in `/etc/os-release` and the Arch packaging

```
==> Starting build()...
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PROJECT_VERSION = 1.8.0
-- BUILD_NUMBER = dev-0
-- BUILD_VERSION = npu-linux-driver-dev-1.8.0.dev-0
-- CMAKE_BUILD_TYPE = RelWithDebInfo
-- option ENABLE_OPENVINO_PACKAGE: OFF
-- option ENABLE_NPU_COMPILER_BUILD: OFF
-- ID = arch
CMake Warning at cmake/detect_linux_system.cmake:21 (message):
  Could not find 'VERSION_ID=.*' pattern in /etc/os-release
Call Stack (most recent call first):
  cmake/detect_linux_system.cmake:33 (read_os_release)
  CMakeLists.txt:45 (include)

CMake Error at cmake/detect_linux_system.cmake:24 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  cmake/detect_linux_system.cmake:33 (read_os_release)
  CMakeLists.txt:45 (include)

-- ENABLE_SANITIZER not specified, using no sanitizers
-- option ENABLE_SANITIZER: 
-- Found LevelZero: /usr/include
-- LevelZero_LIBRARIES: /usr/lib/libze_loader.sodl
-- LevelZero_INCLUDE_DIRS: /usr/include
-- Found Python3: /usr/bin/python3.12 (found version "3.12.7") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Level Zero driver version: 1729356592
-- Found OpenSSL: /usr/lib/libcrypto.so (found version "3.3.2")
CMake Error at cmake/packaging/setup.cmake:45 (message):
  Unable to detect package type for this system
Call Stack (most recent call first):
  CMakeLists.txt:55 (include)

-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
```

```
$ cat /etc/os-release
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo
```

Building from this pkgbuild https://github.com/FabioLolix/PKGBUILD-WIP/blob/main/cauldron/intel-npu-driver-git/PKGBUILD

jwludzik, 2024-10-25:
Sorry for this inconvenient. Feel free to change this line of code:
https://github.com/intel/linux-npu-driver/blob/main/cmake/packaging/setup.cmake#L45 
to 
```
    return()
```

mateusztabaka, 2024-11-08:
@FabioLolix can you pull the main branch now and see that resolves your issue https://github.com/intel/linux-npu-driver/blob/main/cmake/packaging/setup.cmake#L45-L46 ?

xanderlent, 2024-11-09:
Thanks for making the CPack failure non-fatal! This removes the need to patch it out downstream when writing packages for other Linux distributions, like Fedora.

FabioLolix, 2024-11-10:
@mateusztabaka that is resolved now :)

Other issues are:
*  Lack of `VERSION_ID=` in `/etc/os-release` is a build stopper, I have edit it locally but this isn't an option when building in clean chroot with Arch Linux tools
* Can't find `drm/drm.h` 

```
[ 43%] Linking CXX static library libnpu_elf.a
In file included from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/memory/vpu_buffer_object.hpp:14,
                 from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/memory/vpu_buffer_object.cpp:8:
/home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/linux/include/uapi/drm/ivpu_accel.h:10:10: fatal error: drm/drm.h: No such file or directory
   10 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [umd/vpu_driver/source/CMakeFiles/vpu_driver.dir/build.make:261: umd/vpu_driver/source/CMakeFiles/vpu_driver.dir/memory/vpu_buffer_object.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/os_interface/vpu_driver_api.hpp:20,
                 from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/os_interface/vpu_driver_api.cpp:8:
/home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/linux/include/uapi/drm/ivpu_accel.h:10:10: fatal error: drm/drm.h: No such file or directory
   10 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [umd/vpu_driver/source/CMakeFiles/vpu_driver.dir/build.make:289: umd/vpu_driver/source/CMakeFiles/vpu_driver.dir/os_interface/vpu_driver_api.cpp.o] Error 1
In file included from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/memory/vpu_buffer_object.hpp:14,
                 from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/device/vpu_device_context.hpp:15,
                 from /home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/umd/vpu_driver/source/command/vpu_query_command.cpp:9:
/home/fabio/Dev/Github/PKGBUILD-WIP/cauldron/intel-npu-driver-git/src/linux-npu-driver/linux/include/uapi/drm/ivpu_accel.h:10:10: fatal error: drm/drm.h: No such file or directory
   10 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
[...]
```

Which is in:
```
$ locate  drm/drm.h
/usr/include/libdrm/drm.h
/usr/lib/modules/6.11.6-arch1-1/build/include/uapi/drm/drm.h
```

mateusztabaka, 2024-11-12:
@FabioLolix 
First issue:
> Lack of VERSION_ID= in /etc/os-release is a build stopper, I have edit it locally but this isn't an option when building in clean chroot with Arch Linux tools

Can you pull from https://github.com/mateusztabaka/linux-npu-driver/tree/version_id and see if that resolves the problem?

Second issue:
> Can't find drm/drm.h

On Ubuntu, this path comes from linux-libc-dev package. I guess the Arch's counterpart is linux-api-headers, but as I can see drm headers are explicitly removed from the package https://gitlab.archlinux.org/archlinux/packaging/packages/linux-api-headers/-/blob/main/PKGBUILD?ref_type=heads#L33 in favor of headers from libdrm.
Do you know why is that? I think /usr/include/drm/drm.h and /usr/include/libdrm/drm.h can coexist.

But for now, as a test can you give the following workaround a try: when you build the package, can you patch the sources with this https://github.com/mateusztabaka/linux-npu-driver/commit/4be7b84f1d860e2fcf1c69673bce04ff43433d48?
Just want to see if that unblocks you or if there's something else to be resolved.

mateusztabaka, 2024-11-18:
@FabioLolix, fix for below issue was merged to main
> Lack of VERSION_ID= in /etc/os-release is a build stopper, I have edit it locally but this isn't an option when building in clean chroot with Arch Linux tools

So hopefully you don't need the patch anymore in your PKGBUILD.

FabioLolix, 2024-11-18:
Many thanks for the patches!

>  I can see drm headers are explicitly removed from the package https://gitlab.archlinux.org/archlinux/packaging/packages/linux-api-headers/-/blob/main/PKGBUILD?ref_type=heads#L33 in favor of headers from libdrm.
> Do you know why is that?

I have no idea; first time encountering this

I had a last issue with FORTIFY_SOURCE=, Arch set it to 3 and it was failing but found a way to set ti to 2 https://github.com/FabioLolix/PKGBUILD-WIP/commit/aea5424be27e69bd4983d55121b820cf3319f7c6

closed 2024-11-18 by FabioLolix