intel/linux-npu-driver #64: Debian Core Ultra 7 265K Arrow Lake

adoreparler, 2024-12-23:
Just to preface this post, I am a complete noob with the NPU/AI stuff, so starting to try and figure this all out.

I have a new Core Ultra 7 265K with Debian Testing installed on it. 

I would like to start seeing if I can leverage the Intel NPU for LLM stuff (Whisper.cpp) I am hoping I can get it all to work after getting this driver to work and Openvino and then whisper.cpp.

Sometimes the packages for Ubuntu work with Debian so the first think I tried was downloading the packages and trying to install them.
```
wget https://github.com/intel/linux-npu-driver/releases/download/v1.10.1/intel-driver-compiler-npu_1.10.1.20241220-12430270326_ubuntu24.04_amd64.deb
wget https://github.com/intel/linux-npu-driver/releases/download/v1.10.1/intel-fw-npu_1.10.1.20241220-12430270326_ubuntu24.04_amd64.deb
wget https://github.com/intel/linux-npu-driver/releases/download/v1.10.1/intel-level-zero-npu_1.10.1.20241220-12430270326_ubuntu24.04_amd64.deb
wget https://github.com/oneapi-src/level-zero/releases/download/v1.17.44/level-zero_1.17.44+u22.04_amd64.deb
```

```
dpkg -i *.deb
```

and then rebooted

There was nothing at ls /dev/accel/accel0 or ls /dev/accel

```
modinfo intel_vpu
```
Brings back ```modinfo: ERROR: Module intel_vpu not found.```

Do I need build a new kernel with intel_vpu configured in, or is there a module that can be loaded (or is that this driver?), 

I am little lost :/

jwludzik, 2025-01-02:
I am sorry, I am not too familiar with Debian testing. Best would be to check if the kernel config set the CONFIG_DRM_ACCEL_IVPU to y or m, ex. here is a part of my Ubuntu config: (from file /boot/config-6.5.0-44-generic)
```
...
CONFIG_DRM_ACCEL=y
CONFIG_DRM_ACCEL_HABANALABS=m
CONFIG_DRM_ACCEL_IVPU=m
...
```
In Debian kernel config might be in other place.

Can you provide the version of your kernel? Can you check your kernel config for value of CONFIG_DRM_ACCEL*? Thanks in advance

adoreparler, 2025-01-02:
Kernel Version: 6.11.10-amd64

```
cat /boot/config-$(uname -r) |grep CONFIG_DRM_ACCEL
# CONFIG_DRM_ACCEL is not set
```

jwludzik, 2025-01-02:
I am unable to find any fast and easy option. I can only propose two solutions today:
* recompile the actual Debian kernel with CONFIG_DRM_ACCEL_IVPU set to `m` (as module). This should allow you to run `modinfo intel_vpu`. See `Enable the intel_vpu module in the kernel config for compilation` in [docs/overview.md](https://github.com/intel/linux-npu-driver/blob/main/docs/overview.md#kernel-module-driver). I found [KernelFAQ](https://wiki.debian.org/KernelFAQ) in Debian wiki that explains how to recompile kernel
* more unstable option is to install the kernel from [Ubuntu Kernel Mainline](https://kernel.ubuntu.com/mainline/v6.11.11/). I do not guarantee that all Debian packages will work with this kernel

adoreparler, 2025-01-02:
Actually... I have not rebooted in a while and I just did and it loaded the newer kernel that I upgraded to a while back lol
So I am not on 6.11.10-amd64, I am on 6.12.6-amd64
and 
```
➜  ~ cat /boot/config-$(uname -r) |grep CONFIG_DRM_ACCEL
CONFIG_DRM_ACCEL=y
CONFIG_DRM_ACCEL_HABANALABS=m
CONFIG_DRM_ACCEL_IVPU=m
# CONFIG_DRM_ACCEL_QAIC is not set
```

```
➜  ~ modinfo intel_vpu
filename:       /lib/modules/6.12.6-amd64/kernel/drivers/accel/ivpu/intel_vpu.ko.xz
version:        1.0.
license:        GPL and additional rights
description:    Driver for Intel NPU (Neural Processing Unit)
author:         Intel Corporation
firmware:       intel/vpu/vpu_40xx_v0.0.bin
firmware:       intel/vpu/vpu_37xx_v0.0.bin
srcversion:     AA869092EC0F2734E02501A
alias:          pci:v00008086d0000643Esv*sd*bc*sc*i*
alias:          pci:v00008086d0000AD1Dsv*sd*bc*sc*i*
alias:          pci:v00008086d00007D1Dsv*sd*bc*sc*i*
depends:        drm,drm_shmem_helper
intree:         Y
name:           intel_vpu
retpoline:      Y
vermagic:       6.12.6-amd64 SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        {removed}
sig_hashalgo:   sha256
signature:      {removed}
parm:           disable_recovery:Disables recovery when NPU hang is detected (bool)
parm:           tdr_timeout_ms:Timeout for device hang detection, in milliseconds, 0 - default (ulong)
parm:           ivpu_log_level:NPU firmware default trace level: debug=1 info=2 warn=3 error=4 fatal=5 (uint)
parm:           firmware:NPU firmware binary in /lib/firmware/.. (charp)
parm:           dbg_mask:Driver debug mask. See IVPU_DBG_* macros. (int)
parm:           test_mode:Test mode mask. See IVPU_TEST_MODE_* macros. (int)
parm:           pll_min_ratio:Minimum PLL ratio used to set NPU frequency (byte)
parm:           pll_max_ratio:Maximum PLL ratio used to set NPU frequency (byte)
parm:           sched_mode:Scheduler mode: 0 - Default scheduler, 1 - Force HW scheduler (int)
parm:           disable_mmu_cont_pages:Disable MMU contiguous pages optimization (bool)
parm:           force_snoop:Force snooping for NPU host memory access (bool)
```

closed 2025-01-02 by adoreparler