intel/linux-npu-driver #65: Wrong install dir for libze_intel_vpu.so in Ubuntu24.04, Kernel version 6.8.0

Kepontry, 2024-12-26:
By using `sudo cmake --install build --prefix /usr` to install the compiled libze_intel_vpu.so, the dynamic library is installed to the dir `/usr/lib`. However, in my environment, the dir `/usr/lib/x86_64-linux-gnu/` is the right location(I modified several `printf` code and verify that). Do we need to fix that?

jwludzik, 2025-01-02:
You are correct, it looks like a bug in documentation. As a quick workaround I would suggest to drop `--prefix /usr` and let the CMake to choose default path, ex. in my Ubuntu cmake uses `/usr/local/lib`.
```
$ sudo cmake --install build
-- Install configuration: "Release"
-- Up-to-date: /lib/firmware/updates/intel/vpu/mtl_vpu_v0.0.bin
-- Up-to-date: /lib/firmware/updates/intel/vpu/vpu_37xx_v0.0.bin
-- Up-to-date: /lib/firmware/updates/intel/vpu/vpu_40xx_v0.0.bin
...
-- Installing: /usr/local/lib/libze_intel_vpu.so.1.10.1
-- Installing: /usr/local/lib/libze_intel_vpu.so.1
-- Installing: /usr/local/lib/libze_intel_vpu.so
...
```

I remember that I added `--prefix /usr` because I had some trouble with installing the firmware in correct location. It should be fixed, but I might miss to adjust the documents. I will investigate it further. Thanks for reporting an issue!

jwludzik, 2025-01-02:
One more thought on the issue. The `/usr/lib` is a correct path in my case, `/usr/lib` is equal to `/lib` in Ubuntu. Sometimes when library is not found by linker, you need to reload the bindings by using `sudo ldconfig`

jwludzik, 2025-02-19:
Closing issue, did not receive any feedback from main author of the issue. In case of doubt, please reopen

closed 2025-02-19 by jwludzik