intel/linux-npu-driver #66: debug libnpu_driver_compiler is not copied to "build/lib" and build error  reported

zxue2, 2025-01-09:
Steps as below. Please have a check. 

(1) Pull npu driver according to  [[linux-npu-driver/docs/overview.md at main · intel/linux-npu-driver](https://github.com/intel/linux-npu-driver/blob/main/docs/overview.md)](https://github.com/intel/linux-npu-driver/blob/main/docs/overview.md)

(2）Change npu_compiler_build.cmake. 

diff --git a/compiler/npu_compiler_build.cmake b/compiler/npu_compiler_build.cmake
index d61018f..9941dd0 100644
--- a/compiler/npu_compiler_build.cmake
+++ b/compiler/npu_compiler_build.cmake
@@ -27,7 +27,9 @@ list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_TOOLCHAIN_FILE:FILEPATH=${CMAKE_TOO
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_MAKE_PROGRAM:FILEPATH=${CMAKE_MAKE_PROGRAM})
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DBUILD_COMPILER_FOR_DRIVER=ON)
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DBUILD_SHARED_LIBS=OFF)
-list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_BUILD_TYPE=Release)
+list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_BUILD_TYPE=Debug)
+#list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_CXX_FLAGS_DEBUG="-O1")
+list(APPEND NPU_COMPILER_CMAKE_FLAGS -DCMAKE_CXX_FLAGS_DEBUG="-g")
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DENABLE_CLANG_FORMAT=OFF)
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DENABLE_GAPI_PREPROCESSING=OFF)
 list(APPEND NPU_COMPILER_CMAKE_FLAGS -DENABLE_HETERO=OFF)

 
(3) Compile npu driver with compiler as below

```
cmake -B build -S .  -DENABLE_NPU_COMPILER_BUILD=ON -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS_DEBUG="-g -O0"

cmake --build build --parallel  $(nproc)   
```

(4) Error compilation log snippet

[100%] **Linking CXX shared library ../../../../../../bin/intel64/Debug/libnpu_driver_compiler.so**
[100%] Built target npu_driver_compiler
[ 65%] Performing install step for 'npu_compiler_build'
**cp: cannot stat '/root/npu/linux-npu-driver/build/third_party/npu_plugin/build-cid/bin/intel64/Release/libnpu_driver_compiler.so': No sule or directory**
gmake[2]: *** [compiler/CMakeFiles/npu_compiler_build.dir/build.make:105: third_party/openvino/src/npu_compiler_build-stamp/npu_compilerd-install] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1375: compiler/CMakeFiles/npu_compiler_build.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2

(4) Find the debug lib  under linux-npu-driver/build/third_party/npu_plugin/build-cid/bin/intel64/Debug

jwludzik, 2025-01-09:
You need to change "Release" to "Debug" in the https://github.com/intel/linux-npu-driver/blob/93fb54b9d42e7f0f6590f9134aaac92bbf226909/compiler/npu_compiler_build.cmake#L73

zxue2, 2025-01-13:
Thanks Jozef.   It would be great if how to enable debug info for compiler can be added to README or somewhere, because we can't simply enable it by  adding CMAKE_BUILD_TYPE & CMAKE_CXX_FLAGS_DEBUG  in the cmake build command directly.