intel/linux-npu-driver #69: Level-zero Package for Ubuntu 24.04

ekurniaw, 2025-01-23:
On the level-zero package installation, the release document only install for Ubuntu 22.04 (`level-zero_1.17.44+u22.04_amd64.deb`). Should we use 24.04 version (level-zero_1.17.44+u24.04_amd64.deb) for the Ubuntu 24.04?
https://github.com/intel/linux-npu-driver/releases/tag/v1.10.1

Also, is the package necessary? If we do `apt install` after the `dpkg` installation, `apt` will remove the package.
```
$ sudo apt -y install curl
...
(Noting disappearance of level-zero, which has been completely replaced.)
...
The following package disappeared from your system as
all files have been overwritten by other packages:
  level-zero
Note: This is done automatically and on purpose by dpkg.
```

jwludzik, 2025-01-27:
level-zero is required for the user app (ex. OpenVINO) to use the NPU through Level Zero API. Linux NPU driver only implements Level Zero API. If the level-zero package is not present, then the Linux NPU driver is useless.

I know that level-zero might be present in different version then the one recommend in release page, because the GPU solution requires level-zero as well in specific version.

ekurniaw, 2025-01-31:
Yes. Might have used other level-zero. Otherwise, the one for Ubuntu 22.04 is not compatible with Ubuntu 24.04.

closed 2025-01-31 by ekurniaw