intel/linux-npu-driver #77: Build errors on LNL Ubuntu 24.10

saananth, 2025-03-05:
I'm trying to build the linux-npu-driver on the lunar lake AI PC dev kit. I'm running into errors related triggered by -Werror=maybe-uninitialized. The build arguments are the following:

```
cmake -DENABLE_NPU_COMPILER_BUILD=ON ../
make -j8
```

Here are examples of some of the errors errors:

```
build/compiler/src/npu_compiler/src/vpux_compiler/include/vpux/compiler/dialect/VPURegMapped/descriptors.hpp:311:7:
build/compiler/src/npu_compiler/thirdparty/llvm-project/llvm/include/llvm/ADT/DenseMap.h:992:11: error: ‘((elf::Version*)((char*)this + 8))[5]’ may be used uninitialized [-Werror=maybe-uninitialized]
  992 |           ::new (&RHSB->getSecond()) ValueT(std::move(LHSB->getSecond()));
```

And another:
```
/build/compiler/src/npu_compiler/src/vpux_compiler/include/vpux/compiler/dialect/VPURegMapped/descriptors.hpp:311:7:
/usr/include/c++/14/bits/move.h:221:11: error: ‘((llvm::detail::DenseMapPair<llvm::StringRef, elf::Version>**)this)[9]’ may be used uninitialized [-Werror=maybe-uninitialized]
  221 |       _Tp __tmp = _GLIBCXX_MOVE(__a);
```

Build environment:
Compiler: gcc (Ubuntu 14.2.0-4ubuntu2) 14.2.0
OS: Ubuntu 24.10 oracular
GLIBC: 2.40
kernel:  6.11.0-18-generic

I understand this is coming from the NPU compiler. I'm wondering if I'm missing something.

Thanks
-Sriram

jwludzik, 2025-03-05:
It is possible that using new compiler you run into compiler warning as errors. Compiler code has been tested using compiler available in Ubuntu24.04, not 24.10.

Feel free to disable compiler warning as errors. You can patch the compiler code in the same way as done in https://github.com/intel/linux-npu-driver/pull/76/files#diff-b66bfed577cf9435757ae0792397d856881afd169417ea2122029e91d4bde51c

saananth, 2025-03-06:
@jwludzik Thank you. Disabling the warning as errors fixed the issue.

closed 2025-03-06 by saananth