intel/linux-npu-driver #78: the drivers installation failed

goxia, 2025-03-07:
(正在读取数据库 ... 系统当前共安装有 153236 个文件和目录。)
准备解压 intel-driver-compiler-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb  ...
正在解压 intel-driver-compiler-npu (1.13.0.20250131-13074932693) ...
dpkg-deb (子进程)：无法从 intel-driver-compiler-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb 中拷贝归档成员到解码管道中：未预期的文件结束或流结束
dpkg-deb (子进程)：正在解压缩归档“intel-driver-compiler-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb”（大小=18587648）成员“data.tar”：内部 gzip 读取错误：buffer error
dpkg-deb: 错误: <解压缩> 子进程返回错误状态 2
dpkg: 处理归档 intel-driver-compiler-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb (--install)时出错：
 无法为 './usr/lib/x86_64-linux-gnu/libnpu_driver_compiler.so' 拷贝解压的文件至 '/usr/lib/x86_64-linux-gnu/libnpu_driver_compiler.so.dpkg-new'： 未预期的文件结束或流结束
dpkg-deb: 错误: 在 intel-fw-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb 中读取 归档的魔法版本数 时遇到意料之外的文件结束符
dpkg: 处理归档 intel-fw-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb (--install)时出错：
 dpkg-deb --control 子进程返回错误状态 2
dpkg-deb: 错误: 在 intel-level-zero-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb 中读取 归档的魔法版本数 时遇到意料之外的文件结束符
dpkg: 处理归档 intel-level-zero-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb (--install)时出错：
 dpkg-deb --control 子进程返回错误状态 2
在处理时有错误发生：
 intel-driver-compiler-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb
 intel-fw-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb
 intel-level-zero-npu_1.13.0.20250131-13074932693_ubuntu24.04_amd64.deb

jwludzik, 2025-03-10:
Try to download the packages once more and compare them. It seems like you downloaded packages that are incomplete

jwludzik, 2025-03-24:
Closing, there is no feedback from issuer

closed 2025-03-24 by jwludzik