intel/linux-npu-driver #82: Unable to get NPU driver from new kernel 6.14 working with openvino in python

sr99622, 2025-03-29:
I am able to get the NPU driver working if I install from this repository, however, it doesn't seem to work from the new kernel module. These are the steps I took on a Core ULTRA 5 125h (GMKtec) to observe the results.

<h2>Install From Repository</h2>

```
Install Ubuntu 24.10
sudo apt update
sudo apt upgrade

sudo apt install -y libtbb12
wget https://github.com/intel/linux-npu-driver/releases/download/v1.16.0/intel-driver-compiler-npu_1.16.0.20250328-14132024782_ubuntu24.04_amd64.deb
wget https://github.com/intel/linux-npu-driver/releases/download/v1.16.0/intel-fw-npu_1.16.0.20250328-14132024782_ubuntu24.04_amd64.deb
wget https://github.com/intel/linux-npu-driver/releases/download/v1.16.0/intel-level-zero-npu_1.16.0.20250328-14132024782_ubuntu24.04_amd64.deb
wget https://github.com/oneapi-src/level-zero/releases/download/v1.20.2/level-zero_1.20.2+u22.04_amd64.deb
sudo dpkg -i *.deb
sudo reboot -h now

sudo chown root:render /dev/accel/accel0
sudo chmod g+rw /dev/accel/accel0
sudo usermod -a -G render $USER
sudo bash -c "echo 'SUBSYSTEM==\"accel\", KERNEL==\"accel*\", GROUP=\"render\", MODE=\"0660\"' > /etc/udev/rules.d/10-intel-vpu.rules"
sudo udevadm control --reload-rules
sudo udevadm trigger --subsystem-match=accel
sudo reboot -h now

ls -lah /dev/accel/accel0
crw-rw---- 1 root render 261, 0 Mar 29 14:30 /dev/accel/accel0

sudo apt install virtualenv
virtualenv myenv
source myenv/bin/activate
pip install openvino
python
import openvino
openvino.Core().available_devices
['CPU', 'NPU']
```

<h2>Install New Kernel</h2>

```
Install Ubuntu 24.10
sudo apt update
sudo apt upgrade

wget https://kernel.ubuntu.com/mainline/v6.14/amd64/linux-headers-6.14.0-061400-generic_6.14.0-061400.202503241442_amd64.deb
wget https://kernel.ubuntu.com/mainline/v6.14/amd64/linux-headers-6.14.0-061400_6.14.0-061400.202503241442_all.deb
wget https://kernel.ubuntu.com/mainline/v6.14/amd64/linux-image-unsigned-6.14.0-061400-generic_6.14.0-061400.202503241442_amd64.deb
wget https://kernel.ubuntu.com/mainline/v6.14/amd64/linux-modules-6.14.0-061400-generic_6.14.0-061400.202503241442_amd64.deb
sudo dpkg -i *.deb
sudo reboot -h now
```

Here I check the kernel is installed with the NPU driver, all seems ok

```
uname -r
6.14.0-061400-generic

ls -lah /dev/accel/accel0
crw-rw---- 1 root render 261, 0 Mar 29 13:53 /dev/accel/accel0
```

Now I add myself to the render group as in the repository install

```
sudo chown root:render /dev/accel/accel0
sudo chmod g+rw /dev/accel/accel0
sudo usermod -a -G render $USER
sudo bash -c "echo 'SUBSYSTEM==\"accel\", KERNEL==\"accel*\", GROUP=\"render\", MODE=\"0660\"' > /etc/udev/rules.d/10-intel-vpu.rules"
sudo udevadm control --reload-rules
sudo udevadm trigger --subsystem-match=accel
sudo reboot -h now
```

But if I create a virtual environment and install openvino, only the CPU is detected

```
sudo apt install virtualenv
virtualenv myenv
source myenv/bin/activate
pip install openvino
python
import openvino
openvino.Core().available_devices
['CPU']
```

Additional steps I took to see if something was missing, I noticed that level-zero was not there, but installing did not help

```
# Check if Level Zero is installed
dpkg -l level-zero
dpkg-query: no packages found matching level-zero

wget https://github.com/oneapi-src/level-zero/releases/download/v1.20.2/level-zero_1.20.2+u24.04_amd64.deb
sudo dpkg -i level-zero*.deb

sudo chown root:render /dev/accel/accel0
sudo chmod g+rw /dev/accel/accel0

sudo reboot -h now
----------SAME RESULT---------
```

jwludzik, 2025-03-31:
I tried the kernel that you shared and it works with NPU.
```
(npu-env) $ uname -r
6.14.0-performance-20250327-14101507880
(npu-env) $ python -c "import openvino; print(openvino.Core().available_devices)"
['CPU', 'GPU', 'NPU']
```

If the `/dev/accel/accel0` is present and in group `render`, then I would suggest to double check the user permission. You can always try to run a test with root access to verify if device is accessible.

Try to use `strace` to see where the initialization sequence fails. The `strace` can print all system calls, so you can find the status of opening `/dev/accel/accel0` by driver or if linker found a `libze_intel_npu.so` in system
```
$ strace -o log -e trace=openat python -c "import openvino; print(openvino.Core().available_devices)"
['CPU', 'GPU', 'NPU']
$ cat log  | grep -E "(accel|libze_intel|libnp).* = [1-9]"
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_gpu.so.1", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_npu.so.1", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/dev/accel/accel0", O_RDWR|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libnpu_driver_compiler.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/dev/accel/accel0", O_RDWR|O_CLOEXEC) = 3
openat(AT_FDCWD, "/dev/accel/accel0", O_RDWR|O_CLOEXEC) = 5
```

sr99622, 2025-03-31:
Thank you so much for your response. I guess I misunderstood that NPU support was included in the new kernel based on this article [Intel Preps Nice NPU Driver Improvements For Linux 6.14](https://www.phoronix.com/news/Intel-NPU-Linux-6.14-IVPU).

From the strace I was able to see that the bare bones installation I made of the new kernel does not contain libze_intel_gpu or libnpu_driver_compiler. Those can be added using the .deb files from this repository, but what I was believing was that the NPU drivers would be working out of the box without need for additional libraries with the new kernel.

This is an issue for me because installing the drivers on a non-Ubuntu system is not an established path. There are a couple beta type projects to do this, but having it in the kernel would be so much better.

jwludzik, 2025-07-07:
It is not possible to push "user space" into kernel. The same happens in case of GPU. Despite of having the kernel with GPU driver, you still need to install a MESA or other library that translate user space graphic framework to kernel ioctls

The Linux NPU driver (the user space) is available as snap driver in Ubuntu - https://snapcraft.io/intel-npu-driver and as well as Debian packages in https://github.com/intel/linux-npu-driver/releases 

I am closing this ticket. In case of doubt, please reopen.

sr99622, 2025-07-07:
Thank you so much for your response. I don't know much about kernel programming, so most of these issues are beyond my understanding. I do prefer the NPU driver over the GPU driver as I find it to be more efficient and more stable for long running processes. I have had issues with the GPU driver on the Core Ultra platform with Arc graphics. For some reason, on that particular platform running Linux only, I do get infrequent crashes. I have not experienced any with NPU.

The big stumbling block for NPU is the installation. The requirement to use chown, chmod and usermod from the terminal to get the NPU driver working is overwhelming for the average user. The snap package requires these same steps to get it working. As a note, the snap can be installed on other platforms, such as Fedora, but what I have found is that this arrangement dis-engages the process from the system icon, which is very distracting for regular users.

I hope I don't sound like I'm nitpicking, I just wanted to share some feedback from my experience in the field. Thank you so much for your efforts, they are greatly appreciated.

frenchwr, 2025-07-07:
> As a note, the snap can be installed on other platforms, such as Fedora, but what I have found is that this arrangement dis-engages the process from the system icon, which is very distracting for regular users.

Maintainer of the NPU snap here. Could you elaborate on this? Or even better, would you mind opening an issue at https://github.com/canonical/intel-npu-driver-snap/issues ?

sr99622, 2025-07-07:
@frenchwr , I think I might have been a little more nuanced in my explanation. When installing my application as a snap which includes the NPU driver on Fedora, the runtime icon does not follow the application icon. The system uses a generic python icon. This does not happen Ubuntu. I'm pretty sure that this is a Fedora issue, they are probably are missing some piece of code that preserves the application icon when it is running as a python app. I would doubt very much that the NPU driver itself is causing the issue. I only brought it up because I haven't been able to figure out a way to get the NPU driver working from a flatpak and I thought using snap would be a good alternative, but the icon thing is a real bummer.

frenchwr, 2025-07-07:
> [@frenchwr](https://github.com/frenchwr) , I think I should have been a little more nuanced in my explanation. When installing my application as a snap which includes the NPU driver on Fedora, the runtime icon does not follow the application icon. The system uses a generic python icon. This does not happen Ubuntu. I'm pretty sure that this is a Fedora issue, they are probably are missing some piece of code that preserves the application icon when it is running as a python app. I would doubt very much that the NPU driver itself is causing the issue. I only brought it up because I haven't been able to figure out a way to get the NPU driver working from a flatpak and I thought using snap would be a good alternative, but the icon thing is a real bummer.

Understood - thanks for the details!