intel/linux-npu-driver #83: Current RHEL version support

crunchybugs2000, 2025-03-30:
Hi!  How about RHEL 9 support?  Is there RHEL 10 support planned?

I am able partial functionality by building my own 6.13 kernel?  npu_shared_tests passes, ze_intel_npu_tests fails, but that may be due to an incomplete linux-npu-driver repo build:

```[  FAILED  ] 21 tests, listed below:
[  FAILED  ] CommandListGraphApiTest.givenCallAppendGraphInitializeAndExecuteWithEventGetExpectedResults
[  FAILED  ] CommandListGraphApiTest.resetCommandListAfterGraphInitThenAppendingGraphExecAndExecuteReturnsSuccess
[  FAILED  ] GraphNativeTest.whenCreatingGraphFromMalformedBufferInvalidArgumentIsReturned
[  FAILED  ] GraphNativeTest.whenCallgetNativeBinaryWithoutSizePointerExpectInvalidNullPointerError
[  FAILED  ] GraphNativeTest.whenCallgetNativeBinaryWithAndWithoutGraphNativeBinaryPointerExpectSuccess
[  FAILED  ] GraphNativeTest.whenCallsetArgumentValueWithInvalidArgumentErrorIsReturned
[  FAILED  ] GraphNativeTest.givenCallgetPropertiesSuccessfullyParsesGraphInformation
[  FAILED  ] GraphNativeTest.whenCallgetArgumentPropertiesSuccessIsReturning
[  FAILED  ] GraphNativeTest.wrongUserInputShouldBeHandled
[  FAILED  ] GraphNativeTest.whenCallgetArgumentProperties2ExpectSuccess
[  FAILED  ] GraphNativeTest.whenCallgetArgumentProperties3ExpectSuccess
[  FAILED  ] CompilerInDriver.versionCheck
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithNullInputReturnsFailure
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithInputSizeZeroReturnsFailure
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithNullBuildOptionsReturnsError
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithNoBuildOptionsReturnsError
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithBuildOptionsReturnsSuccess
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithPerfCountConfigReturnsSuccess
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithEnableProfilingFlagReturnsSuccess
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithEnableProfilingFlagAndLogLevelConfigReturnsSuccess
[  FAILED  ] CompilerInDriver.creatingNgraphLiteWithEnableProfilingFlagAndPerfCountConfigReturnsSuccess
```
npu-kmd-test fails, but may also be due to an incomplete build:

```
/home/me/src/linux-npu-driver/validation/kmd-test/kmd_test.cpp:256: Failure
Expected: (fd) > (-1), actual: -1 vs -1
open() failed with error 2 - No such file or directory

/home/me/src/linux-npu-driver/validation/kmd-test/kmd_test.cpp:473: Failure
Expected: (context.get_major_id()) > (0), actual: 0 vs 0

unknown file: Failure
C++ exception with description "filesystem error: status: Permission denied [/sys/kernel/debug/accel/]" thrown in TearDown().
```

When I try to compile the linux-npu-driver it fails:

```[ 89%] Building CXX object validation/umd-test/CMakeFiles/npu-umd-test.dir/test_external_memory_opencl.cpp.o
/home/me/src/linux-npu-driver/validation/umd-test/test_external_memory_opencl.cpp:19:10: fatal error: CL/cl.h: No such file or directory
   19 | #include <CL/cl.h>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [validation/umd-test/CMakeFiles/npu-umd-test.dir/build.make:496: validation/umd-test/CMakeFiles/npu-umd-test.dir/test_external_memory_opencl.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2700: validation/umd-test/CMakeFiles/npu-umd-test.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

jwludzik, 2025-03-31:
I am unable to response about RHEL plans for linux-npu-driver. Kernel 6.13 might work, not sure about the all the changes placed there

1. The `ze_intel_npu_tests` fails because you are missing a compiler. You can compile it on your own using `-DENABLE_NPU_COMPILER_BUILD=ON` flag. See [docs/overview.md#building-a-driver-together-with-the-compiler](https://github.com/intel/linux-npu-driver/blob/main/docs/overview.md#building-a-driver-together-with-the-compiler)

2. `npu-kmd-test` should work if your kernel has the `intel_vpu` as module. The `npu-kmd-test` should be run as `root` as it uses sysfs and debufs to validate device functionality

3. It means that OpenCL is installed in system, because the CMake is able to detect it. The failure means that compiler is not able to find the OpenCL headers. You can exclude the OpenCL from building manually by removing the https://github.com/intel/linux-npu-driver/blob/3997a44e37c28449757dd939b9985c2f5fc93172/validation/umd-test/CMakeLists.txt#L16

jwludzik, 2025-04-01:
About the OpenCL, there is some problem with it. The driver detects OpenCL in OpenVINO repository that is cloned together with NPU compiler repository. Debugging this issue further

jwludzik, 2025-04-02:
The fix for OpenCL detection has been merged : https://github.com/intel/linux-npu-driver/commit/ca59fc210cccc82dc2ac059c0fd2590a000d2fec

crunchybugs2000, 2025-04-03:
Thanks for your attention on this topic.

Using kernel 6.13.8 with the NPU driver enabled (in the kernel build) just required that I copy the NPU firmware into /lib/firmware/intel/vpu which addresses my primary issue.  I'll try the merged fix to see if I can get a complete build.

crunchybugs2000, 2025-04-17:
Update.

I managed to build and install with the fix incorporated.

Thanks again!

jwludzik, 2025-07-07:
Closing issue because it has been resolved

closed 2025-07-07 by jwludzik