intel/linux-npu-driver #86: Will it support Intel Corporation Alder Lake-N

pippo73, 2025-04-06:
Hello, 
I know it is a low end chipset, but I would like to know if it will support Intel Corporation Alder Lake-N gpu?

Tnx

xanderlent, 2025-04-07:
Hi @pippo73, this driver is specifically for the NPU device ("Intel AI Boost") found in Intel's recent parts. The NPU is distinct hardware block: **it is not a GPU**. Moreover, as far as I'm aware, Alder Lake-N parts **don't have an NPU device compatible with this driver**. (Compare the Ark pages for the [i3-N305](https://www.intel.com/content/www/us/en/products/sku/231805/intel-core-i3n305-processor-6m-cache-up-to-3-80-ghz/specifications.html) with the [Ultra 7 155UL](https://www.intel.com/content/www/us/en/products/sku/239787/intel-core-ultra-7-processor-155ul-12m-cache-up-to-4-80-ghz/specifications.html).)

Those parts do have its predecessor, the older and now-deprecated Intel GNA device, but there is presently no official Linux support for that hardware. Intel published some draft code but stopped working on it after the deprecation; This NPU code seems to be written by an entirely different team of engineers.

pippo73, 2025-04-07:
> Hi [@pippo73](https://github.com/pippo73), this driver is specifically for the NPU device ("Intel AI Boost") found in Intel's recent parts. The NPU is distinct hardware block: **it is not a GPU**. [...]

Thank you for the detailed reply  😊
😢 for the content of the message