intel/linux-npu-driver #87: Issue with NPU recovery from Linux NPU Driver v1.13.0 release

jwludzik, 2025-04-08:
### Issue ###
We let a bug into **v1.13.0** release where the intel_vpu kernel module driver is unable to recover NPU from broken workload because of unhandled error code from firmware. The issue might only occur when user runs inference that ends with **ZE_RESUTL_ERROR_DEVICE_LOST** return code

### Solutions ###
1. Reload the driver to recover NPU
```
sudo rmmod intel_vpu
sudo modprobe intel_vpu
```
2. Or patch the kernel - https://lore.kernel.org/stable/20250401110807.415197-1-jacek.lawrynowicz@linux.intel.com/T/#m8426e2b416b4cfaee9fcbb5d344de5306b68265b 
3. Or use "intel_vpu" from drm-tip - https://gitlab.freedesktop.org/drm/tip
4. Or use older firmware release than v1.13.0

### Additional ### 
In case of repeating ZE_RESUTL_ERROR_DEVICE_LOST error code from inference run despite the applying above workarounds, the model might not be supported by NPU. Please, check issues in OpenVINO repository - https://github.com/openvinotoolkit/openvino/issues 

In the meantime my team works to patch the kernel. Sorry for inconvenience

koush, 2025-04-11:
I assume you mean the bug was let into [v1.16.0](https://github.com/intel/linux-npu-driver/releases/tag/v1.16.0)

jwludzik, 2025-04-14:
The issue is in both versions - v1.13.0, v1.16.0 and every new release. The proposed solution is to patch the kernel or use the latest "intel_vpu" from drm-tip. The vanilla kernel patching is in progress

jwludzik, 2025-07-17:
The kernels that has the backported patch:

- Starting from Vanilla Kernel 6.12 - [bccc7df3cea064d54d16ab71a13dccc9d2ee7b69](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/accel/ivpu?h=v6.12.38&id=bccc7df3cea064d54d16ab71a13dccc9d2ee7b69)
- In Ubuntu Noble 24.04 HWE image 6.14-6.14.0-24.24_24.04.1

The Ubuntu22.04 is still affected