intel/linux-npu-driver #93: Firmware: Failed to Load in Debian

raniaamina, 2025-05-02:
Hi, i just try to install deb for Ubuntu 24.04 in Debian, i checked the package and it look like no conflict with other packages, so i decided to install it.

All packages installed succesfully, but i found this log in dmseg;

```bash
$ sudo dmesg | grep intel_vpu
[    4.441778] intel_vpu 0000:00:0b.0: enabling device (0000 -> 0002)
[    4.445538] intel_vpu 0000:00:0b.0: firmware: failed to load vpu_37xx.bin (-2)
[    4.445548] intel_vpu 0000:00:0b.0: firmware: failed to load vpu_37xx.bin (-2)
[    4.445554] intel_vpu 0000:00:0b.0: firmware: failed to load vpu_37xx.bin (-2)
[    4.452181] intel_vpu 0000:00:0b.0: [drm] Firmware: intel/vpu/vpu_37xx_v0.0.bin, version: 20250306*MTL_CLIENT_SILICON-release*1130*ci_tag_ud202512_vpu_rc_20250306_1130*5064b5debc3
[    4.535283] [drm] Initialized intel_vpu 1.0.0 for 0000:00:0b.0 on minor 0
```

Please help me to understand this error; i check with [Resource monitor](https://github.com/nokyan/resources), it look like NPU still not detected.

[image]

My laptop use Intel(R) Core(TM) Ultra 7 255HX

```bash

$ lspci | grep -i npu        
00:0b.0 Processing accelerators: Intel Corporation Arrow Lake NPU (rev 01)
```

Thanks.

jwludzik, 2025-05-05:
Everything is fine. The `intel_vpu` kernel module tried first firmware with different name. In the end it loaded a correct one: 
```
[drm] Firmware: intel/vpu/vpu_37xx_v0.0.bin
```

The "Resources" capability to display NPU utilization depends on Kernel version. The utilization is available starting from 6.11

raniaamina, 2025-05-05:
@jwludzik Thanks for confirm it; 

this is kernel version i used;

```bash
$ uname -a
Linux msi-vector 6.12.22-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.22-1 (2025-04-10) x86_64 GNU/Linux

```

raniaamina, 2025-05-06:
I've try to run openvino and yeps get some line chart up down, so it works.

I recently got update for some intel firmware from Debian repository, and when i open Resource again; the NPU name and vendor detected as in this screenshot.

[image]

So, it think this issue can be closed. 
Thanks!

closed 2025-05-06 by raniaamina