intel/linux-npu-driver #96: Can you publish a NPU programming guide?

alanzhai219, 2025-05-17:
I did not find any NPU whitepaper or programming guide like api, compiler. Can you provide it?

alanzhai219, 2025-05-19:
@jwludzik Thanks. I do not want to install OpenVINO for its heavy memory cost. Compared to out-of-box toolkit, it is also necessary to teach developers to code. Out-of-box SDK cannot catch the best developers, who need more programming space.

jwludzik, 2025-05-22:
The OpenVINO is not heavy memory cost, it is up to user to set it up. The Linux NPU driver is unusable without OpenVINO today. The libnpu_driver_compiler.so links to OpenVINO (statically), because compiler only supports OpenVINO IR Format

alanzhai219, 2025-08-31:
@adrianboguszewski I think npu is a standalone ip. If I want to enable it in different frameworks or projects, it means openvino is needed for now. It is dependent.
For me or other developers, we have no motivation. NPU toolkit limits external development contributes.
I think most development is from intel internal, not external world.