intel/linux-npu-driver #107: NPU compilation fails when drivers upgrade from v1.13.0 to newer ones

Kepontry, 2025-08-28:
Hi,

I've found that core.compile_model fails on NPU after I upgrade the driver. Everything works correctly on version v1.13.0.

I used the following commands to perform the upgrade:

```
# 1. Uninstall old driver
sudo dpkg --purge --force-remove-reinstreq intel-driver-compiler-npu intel-fw-npu intel-level-zero-npu

# 2. Install new driver, including the 3 debs in the driver and 1 deb in level-zero
sudo dpkg -i *.deb
```

However, after upgrading, I encounter different issues depending on the version:

On v1.16.0, I get a RuntimeError about the LEVEL0 backend:

```
RuntimeError: Exception from src/inference/src/cpp/core.cpp:109:
Exception from src/inference/src/dev/plugin.cpp:53:
Exception from src/plugins/intel_npu/src/compiler_adapter/include/compiler_adapter_factory.hpp:28:
NPU Compiler Adapter must be used with LEVEL0 backend
```

On v1.22.0 and v1.23.0, the application simply crashes with a segmentation fault:

```
[1]    6476 segmentation fault (core dumped)  python3 run_mha_openvino_new.py
```

Environment Details:

OS: [Ubuntu 24.04]
Kernel: [6.11.10]

OpenVINO: [2025.2]

Kepontry, 2025-08-31:
Update:
The RuntimeError is solved, the reason is that the program link the stale level-zero lib. But the segmentation fault still exists. Here is the call stack.

```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffe5daaa5e8 in vclCompilerCreate () from /usr/lib/x86_64-linux-gnu/libnpu_driver_compiler.so
(gdb) bt
#0  0x00007ffe5daaa5e8 in vclCompilerCreate () from /usr/lib/x86_64-linux-gnu/libnpu_driver_compiler.so
#1  0x00007fff68a92a7a in L0::Compiler::compilerInit (compilerPlatformType=2) at ./umd/level_zero_driver/ext/source/graph/vcl_symbols.hpp:21
#2  0x00007fff68a8a95a in L0::Device::Device (this=this@entry=0x6fdd670, driverHandle=driverHandle@entry=0x6aff7a0, device=std::unique_ptr<VPU::VPUDevice> = {...})
    at ./umd/level_zero_driver/core/source/device/device.cpp:54
#3  0x00007fff68a8be5d in std::make_unique<L0::Device, L0::DriverHandle*, std::unique_ptr<VPU::VPUDevice, std::default_delete<VPU::VPUDevice> > > () at /usr/include/c++/13/bits/unique_ptr.h:1070
#4  L0::DriverHandle::DriverHandle (this=0x6aff7a0, vpuDevices=...) at ./umd/level_zero_driver/core/source/driver/driver_handle.cpp:96
#5  0x00007fff68a8c67a in std::make_unique<L0::DriverHandle, std::vector<std::unique_ptr<VPU::VPUDevice, std::default_delete<VPU::VPUDevice> >, std::allocator<std::unique_ptr<VPU::VPUDevice, std::default_delete<VPU::VPUDevice> > > > > () at /usr/include/c++/13/bits/unique_ptr.h:1070
#6  L0::DriverHandle::create (devices=std::vector of length 0, capacity 0) at ./umd/level_zero_driver/core/source/driver/driver_handle.cpp:109
#7  0x00007fff68a8e638 in operator() (__closure=0x7fffffffbc38) at ./umd/level_zero_driver/core/source/driver/driver.cpp:104
#8  std::__invoke_impl<void, L0::Driver::driverInit(ze_init_flags_t)::<lambda()> > (__f=...) at /usr/include/c++/13/bits/invoke.h:61
#9  std::__invoke<L0::Driver::driverInit(ze_init_flags_t)::<lambda()> > (__fn=...) at /usr/include/c++/13/bits/invoke.h:96
#10 operator() (__closure=<optimized out>) at /usr/include/c++/13/mutex:900
#11 operator() (__closure=0x0) at /usr/include/c++/13/mutex:836
#12 _FUN () at /usr/include/c++/13/mutex:836
#13 0x00007ffff7ca1ed3 in __pthread_once_slow (once_control=0x7fff68aff5a4 <L0::driver+68>, init_routine=0x7fff8c2eb420 <__once_proxy>) at ./nptl/pthread_once.c:116
#14 0x00007fff68a8db5d in __gthread_once (__func=<optimized out>, __once=0x7fff68aff5a4 <L0::driver+68>) at /usr/include/x86_64-linux-gnu/c++/13/bits/gthr-default.h:700
#15 std::call_once<L0::Driver::driverInit(ze_init_flags_t)::<lambda()> > (__f=..., __once=...) at /usr/include/c++/13/mutex:907
#16 L0::Driver::driverInit (this=0x7fff68aff560 <L0::driver>, flags=<optimized out>) at ./umd/level_zero_driver/core/source/driver/driver.cpp:88
#17 0x00007fff68a8e2de in L0::init (flags=<optimized out>) at ./umd/level_zero_driver/core/source/driver/driver.cpp:159
#18 0x00007fff68a76c42 in L0::zeInit (flags=<optimized out>) at ./umd/level_zero_driver/api/ze_driver.cpp:27
#19 0x00007fff040de6f8 in loader::zeInit(unsigned int) () from /usr/lib/x86_64-linux-gnu/libze_loader.so.1
#20 0x00007fff040c7fae in loader::context_t::init_driver(loader::driver_t&, unsigned int, _ze_init_driver_type_desc_t*, _ze_global_dditable_t*, _zes_global_dditable_t*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libze_loader.so.1
#21 0x00007fff040c8bbe in loader::context_t::check_drivers(unsigned int, _ze_init_driver_type_desc_t*, _ze_global_dditable_t*, _zes_global_dditable_t*, bool*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libze_loader.so.1
#22 0x00007fff040dba4d in zelLoaderDriverCheck () from /usr/lib/x86_64-linux-gnu/libze_loader.so.1
```

jwludzik, 2025-09-03:
Can you check if you are using libze_intel_**vpu**.so.1 by mistake? I mean there was an action with renaming libze_intel_vpu to libze_intel_npu correlated with Level Zero Loader starting from v1.16.0 release. 

* Try to look for libze_intel_vpu.so* and remove it. Then repeat your test
```bash
find /usr -name libze_intel_vpu.so*
rm -f <path to libze_intel_vpu>
```

* Or try use `strace` to check if libze_intel_vpu.so.1 is loaded by libze_loader.so.1
```bash
(venv) $ strace -o strace.log --trace=file python -c "from openvino import Core; print(Core().available_devices)"
['CPU', 'GPU', 'NPU']
(venv) $ grep libze_intel_ strace.log
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_gpu.so.1", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_gpu_legacy1.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libze_intel_gpu_legacy1.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libze_intel_gpu_legacy1.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libze_intel_gpu_legacy1.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_vpu.so.1", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_npu.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libze_intel_npu.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libze_intel_npu.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libze_intel_npu.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```
The line: `openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libze_intel_vpu.so.1", O_RDONLY|O_CLOEXEC) = 4` is blocking the usage of libze_intel_npu.so.1 when using newer releases. You have to remove `/lib/x86_64-linux-gnu/libze_intel_vpu.so.1` first

* Or enable warning logs from driver
```bash
export ZE_INTEL_NPU_LOGLEVEL=WARNING
```
and see if there is any suspicious log

closed 2025-09-03 by Kepontry